ros-mobile-robots/diffbot #23: Python versions conflict

Russ76, 2021-02-11:
sudo apt-get install python-catkin-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 python-catkin-tools : Depends: python (< 2.8)
                       Depends: python (>= 2.7)
                       Depends: python-catkin-pkg (>= 0.2.9) but it is not installable
                       Depends: python-osrf-pycommon but it is not installable
                       Depends: python:any (>= 2.6.6-7~)
E: Unable to correct problems, you have held broken packages.

This is what I see after trying to install Catkin tools. I followed all directions exactly up to this point.

fjp, 2021-02-11:
Hi @Russ76 thanks raising this issue. I am sorry about that the docs are not up to date on this. I assumed it would be easier to install catkin-tools in the future when I wrote the docs. However, the current way to do it is using this command:

````
sudo apt-get install python3-osrf-pycommon python3-catkin-tools
````

Please see the [discussion here](https://github.com/catkin/catkin_tools/issues/594#issuecomment-688149976). I hope this will be resolved soon - at least the discussion is pretty active, looking at the latest comment.

Russ76, 2021-02-12:
Thanks, that's fast! In the "Mate" browser, the web page looked different (your site, not Github), I had to get the URL from my laptop and type that into Mate to get the full number of web pages with more instructions. I don't know what happened.

fjp, 2021-02-12:
Yes I need to fix this, the two web pages, [fjp.at/projects/diffbot](https://fjp.at/projects/diffbot/ros-noetic/) (old) and [fjp.at/diffbot](https://fjp.at/diffbot/ros-setup/) are not always up to date with the latest [docs in this github repository](https://github.com/fjp/diffbot/tree/noetic-devel/docs).

I hope it is possible to add a github action to build the mkdocs automatically everytime a push is made - I'll create a new issue for this.

I will close this issue for now, as I hope the command above solves this issue for now. And I guess in the future the standard way, using the ubuntu package, will also work.

closed 2021-02-12 by fjp