ros-mobile-robots/diffbot #30: Initialization problem on real robot: wheels immediately start rotating after bringup

fjp, 2021-03-09:
>  I am using "diffbot_bringup_with_laser" and I wonder why the tracks are turning? Shouldn't it wait for a navigation package to be launched? Move_base isn't running yet, is it?

This is a bug, I encounter the same problem on my robot. I suspect missing zero initializations in the `diffbot_hw_interface.cpp` but couldn't find the exact problem so far and didn't investiage further yet. I do set the joint states and commands to zero [here in the code](https://github.com/fjp/diffbot/blob/e5721816d70b59c9f0374c49914e945c772345a9/diffbot_base/src/diffbot_hw_interface.cpp#L75) at the beginning but it doesn't seem to help.

Another source of this problem might be the [motor driver code](https://github.com/fjp/diffbot/blob/noetic-devel/grove_motor_driver/src/grove_i2c_motor_driver.py).

What I do to "solve" this, is to send an empty message on the `/diffbot/reset` topic to the encoders that might not be zero while the wheels spin. And then in my case I can stop the wheels by just blocking the wheels of the robot during this resetting, or I also make use of the motor driver reset button.

_Originally posted by @fjp in https://github.com/fjp/diffbot/issues/24#issuecomment-794489216_

fjp, 2021-03-20:
<details><summary>Hardware Interface log</summary>
```
fjp@diffbot:~/catkin_ws$ roslaunch diffbot_bringup bringup.launch 
... logging to /home/fjp/.ros/log/82733adc-8954-11eb-a952-438647b07a39/roslaunch-diffbot-17421.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.26:37407/

SUMMARY
========

PARAMETERS
 * /diffbot/hardware_interface/joints: ['front_left_whee...
 * /diffbot/joint_state_controller/extra_joints: [{'name': 'rear_c...
 * /diffbot/joint_state_controller/publish_rate: 50
 * /diffbot/joint_state_controller/type: joint_state_contr...
 * /diffbot/mobile_base_controller/angular/z/has_acceleration_limits: True
 * /diffbot/mobile_base_controller/angular/z/has_velocity_limits: True
 * /diffbot/mobile_base_controller/angular/z/max_acceleration: 6.0
 * /diffbot/mobile_base_controller/angular/z/max_velocity: 2.0
 * /diffbot/mobile_base_controller/base_frame_id: base_footprint
 * /diffbot/mobile_base_controller/left_wheel: front_left_wheel_...
 * /diffbot/mobile_base_controller/linear/x/has_acceleration_limits: True
 * /diffbot/mobile_base_controller/linear/x/has_velocity_limits: True
 * /diffbot/mobile_base_controller/linear/x/max_acceleration: 0.6
 * /diffbot/mobile_base_controller/linear/x/max_velocity: 0.2
 * /diffbot/mobile_base_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /diffbot/mobile_base_controller/publish_rate: 50
 * /diffbot/mobile_base_controller/right_wheel: front_right_wheel...
 * /diffbot/mobile_base_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /diffbot/mobile_base_controller/type: diff_drive_contro...
 * /diffbot/mobile_base_controller/wheel_radius: 0.0325
 * /diffbot/mobile_base_controller/wheel_separation: 0.15
 * /diffbot/robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES
  /diffbot/
    controller_spawner (controller_manager/spawner)
    diffbot_base (diffbot_base/diffbot_base)
    motor_driver (grove_motor_driver/motor_driver.py)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rosserial_teensy (rosserial_python/serial_node.py)

ROS_MASTER_URI=http://192.168.0.9:11311/

process[diffbot/diffbot_base-1]: started with pid [17477]
process[diffbot/controller_spawner-2]: started with pid [17478]
process[diffbot/motor_driver-3]: started with pid [17479]
process[diffbot/rosserial_teensy-4]: started with pid [17481]
[ INFO] [1616228464.266134126]: Waiting for model URDF on the ROS param server at location: /diffbot/diffbot/robot_description
process[diffbot/robot_state_publisher-5]: started with pid [17486]
[ INFO] [1616228464.459879850]: Initializing DiffBot Hardware Interface ...
[ INFO] [1616228464.460056253]: Number of joints: 2
[ INFO] [1616228464.460419242]: pid namespace: pid/left_motor
[ INFO] [1616228464.460576090]: Initialize PID
[ INFO] [1616228464.460676420]: Initializing dynamic reconfigure in namespace /diffbot/pid/left_motor
[ INFO] [1616228464.557533459]: Update PID Gains: F=0.8, P=0.35, I=0.5, D=0.01, out_min=-6.15385, out_max=6.15385
[ INFO] [1616228464.586639267]: Initialized dynamic reconfigure
[ INFO] [1616228464.586828372]: Initialized PID: F=0.8, P=0.35, I=0.5, D=0.01, out_min=-6.15385, out_max=6.15385
[ INFO] [1616228464.586925684]: Update PID output limits: lower=6.15385, upper=-6.15385
[ INFO] [1616228464.587086976]: pid namespace: pid/right_motor
[ INFO] [1616228464.587175251]: Initialize PID
[ INFO] [1616228464.587251823]: Initializing dynamic reconfigure in namespace /diffbot/pid/right_motor
[ INFO] [1616228464.680847326]: Update PID Gains: F=0.8, P=0.35, I=0.5, D=0.01, out_min=-6.15385, out_max=6.15385
[ INFO] [1616228464.706535566]: Initialized dynamic reconfigure
[ INFO] [1616228464.706727079]: Initialized PID: F=0.8, P=0.35, I=0.5, D=0.01, out_min=-6.15385, out_max=6.15385
[... 312 lines cut ...]
j1:       27608     320.048   -439.672  -4397.93  
[ INFO] [1616228467.935462694]: 
Write     velocity  p_error   i_error   d_error   pid out   percent   
j0:       0         -8.19972e+07-8.19747e+06-8.20197e+08-3.6901e+07-5.99641e+08
j1:       0         4397.93   439.672   43991.3   1982.69   32218.7   

[ INFO] [1616228468.034861437]: 
Read      ticks     angle     dangle    velocity  
j0:       -5038     -58.4035  -0.324593 -3.24519  
j1:       27636     320.373   0.324593  3.24519   
[ INFO] [1616228468.035243203]: 
Write     velocity  p_error   i_error   d_error   pid out   percent   
j0:       0         3.24519   -8.19747e+068.19787e+088.19787e+061.33215e+08
j1:       0         -3.24519  439.348   -44001.8  -437.654  -7111.87  

[ INFO] [1616228468.134896549]: 
Read      ticks     angle     dangle    velocity  
j0:       -5024     -58.2412  0.162296  1.62269   
j1:       27621     320.199   -0.173889 -1.73859  
[ INFO] [1616228468.135283612]: 
Write     velocity  p_error   i_error   d_error   pid out   percent   
j0:       0         -1.62269  -8.19747e+06-48.6705  -4.55465  -74.013   
j1:       0         1.73859   439.522   49.8294   4.6068    74.8605
```
</details>

fjp, 2021-03-28:
I think I found the problem and work on it in the [hw_init branch](https://github.com/fjp/diffbot/tree/hw_init). After it is fixed completely I will do a PR to the main noetic-devel branch.

There are/were mainly three errors causing the problem:
- division by zero time period in the first cycle of the control loop start (this is fixed in commit b05f9c0)
- the encoder_ticks_ variable wasn't initialized correctly (fixed by 1232352)
- The node running on the Teensy microcontroller starts to publish encoder values after the hardware interface control loop (read, update write) is already running. This can cause a jump in encoder tick values, e.g. from zero to whatever value is first published from the Teensy. 

For the last unsolved error I think the best way to solve it, is to let the hw intf control loop wait for the encoder messages being published. Another way in solving this, could be to send the `/reset` message from the hardware interface just before the first loop cycle starts. This will lead to zero ticks and therefore no jump.

Russ76, 2021-03-29:
Sounds like you're fixing it, Franz! I'll have to check through my Teensy code when you finish, as mine is controlling the motors, too. (No Grove device.)

commit 3ccba421 (2021-04-12): fix #30: add method to check if encoder ticks are published
commit f4646a8b (2021-04-30): fix #30: add method to check if encoder ticks are published