ros-mobile-robots/diffbot #34: Issues with diffbot_gazebo diffbot_base.launch

zulfiz, 2021-07-01:
Hi, I am trying to get the diffbot controller working in simulation. When I run the diffbot_base.launch file I see the following error:

`RLException: roslaunch file contains multiple nodes named [/diffbot/robot_state_publisher].
Please check all <node> 'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file
`
I can get it to run if change the name of diffbot/robot_state_publisher/ to something else such as diffbot_2/robot_state_publisher. However, I can't see the controller working in simulation. If I try to do a rostopic of encoder ticks or mesured joint states, I don't see any message being published. I see the following warning message:

`WARNING: no messages received and simulated time is active.
Is /clock being published?`

My steps are:
`roslaunch diffbot_base diffbot.launch
roslaunch diffbot_gazebo diffbot_base.launch
`
Could you please guide me, what am I doing wrong. Thank you!

fjp, 2021-07-02:
Hi, thanks for trying the diffbot packages. The diffbot_base.launch from the diffbot_gazebo package isn't well tested and is intended to run together with the real robot hardware. 

https://github.com/ros-mobile-robots/diffbot/blob/e9e74fe1fe1ede8200c607376ad21539526aad2b/diffbot_gazebo/launch/diffbot_base.launch#L2-L4

So just to clarify, you should run `diffbot_base/launch/diffbot.launch` on the real robot hardware and `diffbot_gazebo/launch/diffbot_base.launch` on your development PC.

If you don't have the robot hardware, it is only possible to work with the simulated robot and the ros control gazebo plugin. For this I suggest you follow the examples in the readme, [here in this section](https://github.com/fjp/diffbot#gazebo-simulation-with-ros-control). 

It is unfortunately not possible to simulate the encoder ticks in gazebo. 

I hope this helps, otherwise just let me know.
Thanks.

fjp, 2021-07-13:
I am closing this but feel free to reply if needed.

closed 2021-07-13 by fjp