ros-mobile-robots/diffbot #40: Syntax error compiling base_controller in VSCode+PIO

joeuser846, 2021-10-05:
I've imported the base_controller project directory into VSCode+PIO.  I added the Paul Stoffregen "Encoder" library but when I build I get the following syntax errors.  I'd really appreciate a little guidance on what I did wrong and how to fix it.  The issue seems to have something to do with namespace and scope, but exactly what?  Thanks in advance for any help.

[platformio]
default_envs = teensy31
[env:teensy31]
platform = teensy
board = teensy31
framework = arduino
lib_deps = 
       frankjoshua/Rosserial Arduino Library@^0.9.1
      ./lib
      adafruit/Adafruit Motor Shield V2 Library@^1.0.11
      Wire
      paulstoffregen/Encoder@^1.4.1

Building in release mode
Compiling .pio/build/teensy31/src/main.cpp.o
Compiling .pio/build/teensy31/lib821/SPI/SPI.cpp.o
Compiling .pio/build/teensy31/lib8c6/Ethernet/Dhcp.cpp.o
Compiling .pio/build/teensy31/lib8c6/Ethernet/Dns.cpp.o
In file included from .pio/libdeps/teensy31/lib/base_controller.h:17:0,
                 from src/main.cpp:4:
lib/encoder/encoder.h:39:11: error: 'Encoder' in namespace '::' does not name a type
         ::Encoder encoder;
           ^
lib/encoder/encoder.h: In member function 'int32_t diffbot::Encoder::read()':
lib/encoder/encoder.h:89:40: error: 'encoder' was not declared in this scope
         inline int32_t read() { return encoder.read(); };
                                        ^
lib/encoder/encoder.h: In member function 'void diffbot::Encoder::write(int32_t)':
lib/encoder/encoder.h:97:40: error: 'encoder' was not declared in this scope
         inline void write(int32_t p) { encoder.write(p); };
                                        ^

fjp, 2021-10-05:
Hi @joeuser846, this is indeed strange. I noticed you added `paulstoffregen/Encoder@^1.4.1` explicitly in the `platformio.ini`. As far as I know this shouldn't be required, the `platformio.ini` I use looks like this:

https://github.com/ros-mobile-robots/diffbot/blob/11ac235b474b81226086e7af45f786e5e87ec7e8/diffbot_base/scripts/base_controller/platformio.ini#L11-L22

The reason is that the encoder library should be already integrated in the framework-arduino package.
Can you please try compiling after removing this line: `paulstoffregen/encoder@^1.4.1` and let me know if this works for you?

There might be an issue with this version of the library, which comes up also in [this thread](https://community.platformio.org/t/beginner-who-wants-to-migrate-to-platformio-but-a-lot-of-things-that-i-do-not-understand-my-sketch-works-on-arduino-ide/10371) (although it seems to be not directly the same issue you have):

> - it seems you have the Encoder library installed in the local library storage via the VSCode UI. This is not necessary since the encoder library is already integrated in the framework-arduino package. If you remove it, it should take the fixed version from the Arduino Package.
> - force usage of the updated version; that is, in the platformio.ini in the lib_deps directive, instead of saying Encoder give it the git link https://github.com/PaulStoffregen/Encoder.git

After the program compiles please also make sure to have the Teensyduino GUI installed, explained here: https://ros-mobile-robots.com/teensy-mcu/. Alternatively you could also try the [`teensy-cli` `upload_protocol`](https://docs.platformio.org/en/latest/platforms/teensy.html#uploading-using-cli) without installing the GUI. Would be interesting to know if it works also without installing the GUI, because I have always installed and used the Teensyduino GUI so far.

joeuser846, 2021-10-05:
Hello.  Thanks for your quick reply.  To be sure I had not modified anything by accident I removed my repo and cloned it fresh from github.  Then I just opened the PIO-ready project at the “diffbot/diffbot_base/scripts/base_controller” directory.  I’m attaching a screenshot (below) from VSCode that shows the project file hierarchy and the platformio.ini file exactly as imported.  When I build I still get the same syntax errors as you can see in the screenshot.

I’m running VSCode on a MacBook but I’ve never had any problems with incompatibilities in the past.  My ROS Noetic system is on Ubuntu but I just find it easier to do my PlatformIO embedded projects on the Mac.

fjp, 2021-10-05:
I am sorry you ran into this issue, I didn't expect someone will build on other OS than Ubuntu and that this would cause an issue.

I can confirm that the `base_controller` doesn't compile on Windows either. It is a letter case issue. The problem is that Windows and seemingly macOS are case-insensitive to header file names. Ubuntu is case-sensitive and therefore `"encoder.h"` and `<Encoder.h>` are treated differently only on Ubuntu. For Windows and macOS these are the same header.

The problem can be solved by renaming `encoder.h` (and `encoder.cpp`) to something else. For example, I tried with `encoder_diffbot.h`, which makes the script compile successfully on Windows.

You can try this solution for now. If it works for you, a pull request with a suitable new name for the header is most welcome :-) Otherwise please just let me know if it works and I probably will fix this in the near future.

joeuser846, 2021-10-06:
Thanks!  That fixed it.  Kind of an interesting situation though.  It never occurred to me that a working VSCode+PIO project wouldn’t be portable.  There’s always more to learn I guess.

closed 2021-11-04 by fjp