ros-mobile-robots/diffbot #44: Laser scan translation wrong when robot turn

funnyrabbitvu, 2021-12-10:
https://user-images.githubusercontent.com/48290085/145632117-82781293-bd34-4f18-8462-d6b6662ad1a6.mp4

Hi guys,
I'm facing an issue that when I turn robot around in SLAM mode the laser scan effect was displaying wrong pose/translation, specially, its gonna fit the environment only when robot stop moving (showing in video). I'm thinking about that I got wrong odometry calculation or maybe the laser frequency update was not fast enough. This issue effected to navigation process that robot would turn into recovery mode. I didn't find the answer yet so I hope you guys could help me out.
Many thanks in advanced!

Regards,
rabbit.

fjp, 2021-12-11:
Hi @funnyrabbitvu, thanks for reporting this problem! I am also aware of this issue and like you mention this happens most likely because the odometry is not accurate enough. In simulation (and real world) there are a lot of settings to tweak in order to improve this issue. For the new robot Remo, found in the [`remo_description`](https://github.com/ros-mobile-robots/remo_description) package (which is different to the "old [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description)"), I invested a lot of time to tune some of the following parameters. One of the important ones were to set the wheel base (distance between wheels) and wheel radius correctly. Otherwise the odometry calculations are obviously wrong and lead to the problem you are showing.

- wheel base and radius in [`diffbot_control/config`](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_control/config/diffbot_control.yaml)

https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_control/config/diffbot_control.yaml#L30-L34

- Use correct mass values for the components of your robot ([here](https://github.com/ros-mobile-robots/remo_description/commit/7b01d577d40984aaa5fb520291a6fa5b2368bcf8) I tried to do this for Remo). The mass values in [`diffbot_description/config`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description/config/diffbot) are probably not entirely correct yet. 
- Tune the simulated joint/wheel friction in the robot description (see the changes made [here for example](https://github.com/ros-mobile-robots/remo_description/commit/7b01d577d40984aaa5fb520291a6fa5b2368bcf8)). For [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_description/urdf/include/robot.gazebo.xacro#L20) **this hasn't been done yet**.
- Change [gazebo settings of the simulated world](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_gazebo/worlds/turtlebot3_world.world#L13). For comparision [here is turtlebots world settings](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/noetic-devel/turtlebot3_gazebo/worlds/turtlebot3_world.world#L13) (which should be the same).
- Adapt [amcl parameters](http://wiki.ros.org/amcl#Parameters) to compensate for bad odometry using vision sensors (2D laser in our case). Here, `odom_model_type=diff-corrected`, `odom_alphaX` and `update_min_a` improved the behavior (you can see the commit [here](https://github.com/ros-mobile-robots/diffbot/commit/9e155597e7ce1d0db729822e3e48c04c2128c13c) where I tried tuning these values).  

Adapting these parameters improved the behavior but it is still not perfect as you can see in the following video of Remo twisting around:

https://user-images.githubusercontent.com/1286618/145669976-2c20a232-7f12-42be-b426-dc362efa2042.mp4

Now the thing is, that currently most values are tuned for Remo robot and **not** the "old diffbot model" (found in the [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description)), which you are using. To alleviate this issue for both models, we should tune both models as good as possible, e.g. the gazebo related settings in the `diffbot/remo_description` (for diffbot this hasn't been extensivly done yet) and also use a specific parameter set, depending on which robot is used by the user.

What I also noticed is that in Gazebo simulator the scans are not matching to the obstacles either:

https://user-images.githubusercontent.com/1286618/145671681-5315b8c5-2981-48a5-b46a-749b5d9c5d54.mp4

However, this might be not an issue as it also happens with turtlebot3:

https://user-images.githubusercontent.com/1286618/145672320-fae66885-3d0d-4296-a356-2b82e1b472a2.mp4

Regarding why for turtlebot3 the scans are aligned more accurately with the map, I have only some assumptions, which I want to list here:

- Turtlebot3 uses the [`libgazebo_ros_diff_drive`](http://gazebosim.org/tutorials?tut=ros_gzplugins#DifferentialDrive) gazebo plugin ([here for burger](https://github.com/ROBOTIS-GIT/turtlebot3/blob/c89e366ac6f5690db90d7c74352d8452d146f380/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro#L51)). Instead of that, diffbot and Remo use the [`gazebo_ros_control`](http://gazebosim.org/tutorials/?tut=ros_control) plugin in combination with the [`diff_drive_controller`](http://wiki.ros.org/diff_drive_controller).
- Turtlebot3 uses an imu. A (simulated) imu together with encoder odometry fused together by an EKF might improve the odometry (after tuning everything correctly). Here the official gazebo imu plugins might help or the ones from [hector group](http://wiki.ros.org/hector_gazebo_plugins) together with the [robot_localization](http://wiki.ros.org/robot_localization) EKF. A robot that is also worth looking at is [mir_robot](https://github.com/dfki-ric/mir_robot/blob/80e59ad03259eaef27fd39bf946000ccecda9178/mir_description/urdf/include/imu.gazebo.urdf.xacro) or [husky](https://github.com/husky/husky/blob/f19a0e8f555952bff72d224f94b649202f8b256f/husky_description/urdf/husky.urdf.xacro#L220), both use [`libhector_gazebo_ros_imu`](http://wiki.ros.org/hector_gazebo_plugins). The difference between the [`GazeboRosImuSensor`](http://gazebosim.org/tutorials?tut=ros_gzplugins) is [explained here](https://github.com/tu-darmsta [... cut ...]
- https://answers.ros.org/question/258420/ball-joint-in-urdf/
- https://answers.ros.org/question/307520/what-are-the-steps-to-use-diff_drive_controller-in-a-physical-robot/

Finally, I want to highlight that I would love to see a good/clear solution to this issue, so any other findings you have about this are most welcome. I hope find time soon to add an IMU to the robot and also test it extensively in simulation. So far I just did a quick test with a simulated imu but without a lot of parameter tuning (imu and ekf).

funnyrabbitvu, 2021-12-12:
> Hi @funnyrabbitvu, thanks for reporting this problem! I am also aware of this issue and like you mention this happens most likely because the odometry is not accurate enough. In simulation (and real world) there are a lot of settings to tweak in order to improve this issue. For the new robot Remo, found in the [`remo_description`](https://github.com/ros-mobile-robots/remo_description) package (which is different to the "old [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description)"), I invested a lot of time to tune some of the following parameters. One of the important ones were to set the wheel base (distance between wheels) and wheel radius correctly. Otherwise the odometry calculations are obviously wrong and lead to the problem you are showing.
> 
> * wheel base and radius in [`diffbot_control/config`](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_control/config/diffbot_control.yaml)
> 
> https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_control/config/diffbot_control.yaml#L30-L34
> 
> * Use correct mass values for the components of your robot ([here](https://github.com/ros-mobile-robots/remo_description/commit/7b01d577d40984aaa5fb520291a6fa5b2368bcf8) I tried to do this for Remo). The mass values in [`diffbot_description/config`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description/config/diffbot) are probably not entirely correct yet.
> * Tune the simulated joint/wheel friction in the robot description (see the changes made [here for example](https://github.com/ros-mobile-robots/remo_description/commit/7b01d577d40984aaa5fb520291a6fa5b2368bcf8)). For [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_description/urdf/include/robot.gazebo.xacro#L20) **this hasn't been done yet**.
> * Change [gazebo settings of the simulated world](https://github.com/ros-mobile-robots/diffbot/blob/327f39559a7d7f2a49ea5a9534b941f7777e483b/diffbot_gazebo/worlds/turtlebot3_world.world#L13). For comparision [here is turtlebots world settings](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/blob/noetic-devel/turtlebot3_gazebo/worlds/turtlebot3_world.world#L13) (which should be the same).
> * Adapt [amcl parameters](http://wiki.ros.org/amcl#Parameters) to compensate for bad odometry using vision sensors (2D laser in our case). Here, `odom_model_type=diff-corrected`, `odom_alphaX` and `update_min_a` improved the behavior (you can see the commit [here](https://github.com/ros-mobile-robots/diffbot/commit/9e155597e7ce1d0db729822e3e48c04c2128c13c) where I tried tuning these values).
> 
> Adapting these parameters improved the behavior but it is still not perfect as you can see in the following video of Remo twisting around:
> 
>  remo_twist_issue_wheel_base.mp4 
> Now the thing is, that currently most values are tuned for Remo robot and **not** the "old diffbot model" (found in the [`diffbot_description`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_description)), which you are using. To alleviate this issue for both models, we should tune both models as good as possible, e.g. the gazebo related settings in the `diffbot/remo_description` (for diffbot this hasn't been extensivly done yet) and also use a specific parameter set, depending on which robot is used by the user.
> 
> What I also noticed is that in Gazebo simulator the scans are not matching to the obstacles either:
> 
>  remo_twist_issue_gazebo.mp4 
> However, this might be not an issue as it also happens with turtlebot3:
> 
>  turtlebot3_gazebo_twist_issue.mp4 
> Regarding why for turtlebot3 the scans are aligned more accurately with the map, I have only some assumptions, which I want to list here:
> 
> * Turtlebot3 uses the [`libgazebo_ros_diff_drive`](http://gazebosim.org/tutorials?tut=ros_gzplugins#DifferentialDrive) gazebo plugin ([here for burger](https://github.com/ROBOTIS-GIT/turtlebot3/blob/c89e366ac6f5690db90d7c74352d8452d146f380/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro#L51)). Instead of that, diffbot and Remo use the [`gazebo_ros_control`](http://gazebosim.org/tutorials/?tut=ros_control) plugin in combination with the [`diff_drive_controller`](http://wiki.ros.org/diff_drive_controller).
> * Turtlebot3 uses an imu. A (simulated) imu together with encoder odometry fused together by an EKF might improve the odometry (after tuning everything correctly). Here the official gazebo imu plugins might help or the ones from [hector group](http://wiki.ros.org/hector_gazebo_plugins) together with the [robot_localization](http://wiki.ros.org/robot_localization) EKF. A robot that is also worth looking at is [mir_robot](https://github.com/dfki-ric/mir_robot/blob/80e59ad03259eaef27fd39bf946000ccecda9178/mir_description/urdf/include/imu.gazebo.urdf.xacro) or [husky](https://github.com/husky/husky/blob/f19a0e8f555952bff72d224f94b649202f8b256f/husky_description/urdf/husky.urdf.xacro#L220), both use [`libhector_gazebo_ros_imu`](http://wiki.ros.org/hector_gazebo_plugins). The difference between the [`GazeboRosImuSensor`](http://gazebosim.org/tutorials?tut=ros_gzplugins) is [explained here](https://github.com/tu-darmstadt-ros-pkg/hector_gazebo/issues/50).
> * ... there might be other reasons why the scans align more precisely for turtlebot3 compared to Remo
> 
[... cut ...]
> * https://answers.ros.org/question/307520/what-are-the-steps-to-use-diff_drive_controller-in-a-physical-robot/
> 
> Finally, I want to highlight that I would love to see a good/clear solution to this issue, so any other findings you have about this are most welcome. I hope find time soon to add an IMU to the robot and also test it extensively in simulation. So far I just did a quick test with a simulated imu but without a lot of parameter tuning (imu and ekf).

Many thanks for your awesome support, bro. I'm trying to figure it out with your guideline. I will report my working status.