ros-mobile-robots/diffbot #67: Compile Grove motor driver

Russ76, 2022-11-19:
This task proves to be challenging. Preparing the cmakelists.txt was difficult, and now it wants libraries from the Arduino IDE before it will compile! Must the Arduino IDE be installed on the Raspberry Pi before compiling the Grove motor driver package? I copied over the two Python files that are needed from a different repo and added them to the source folder.

fjp, 2022-11-19:
Which Arduino libraries are you trying to compile?

Last time I used it was with the [GovePi Python lib from DexterInd](https://github.com/DexterInd/GrovePi), please see tag [0.0.1](https://github.com/ros-mobile-robots/diffbot/tree/0.0.1/grove_motor_driver/src)

Russ76, 2022-11-19:
I'm trying to compile the Grove I2C motor driver!

fjp, 2022-11-20:
I guess you referring to the Seeed-Studio [grove motor drive library](https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver_v1_3). I think this one is only for Arduino mcus and not directly suitable for the Raspberry arm platform.

From the [header file](https://github.com/Seeed-Studio/Grove_I2C_Motor_Driver_v1_3/blob/master/Grove_I2C_Motor_Driver.h#L35) it seems so that the Arduino.h needs to be present. So to build it for the Raspberry Pi you would have to provide this header and all it's includes (which are a lot I guess) and then add it to the target_include_headers in your CMakeLists.txt. I also assume it isn't header only and you most likely then have to link against the Arduino libraries (target_link_libraries). I have never tried or seen it before to run Arduino libraries (such as the Grove motor driver from Seeed Studio) on another platform (in this case Raspberry Pi arm platform). So I really don't know if this is possible at all. 

Maybe there is a project that successfully ported Arduino libs to Raspberry Pi. Or even better maybe there is already a Grove motor driver c++ lib for the Raspberry pi.

The Python library from DexterInd that I used for the Grove motor driver was specifically for the Raspberry Pi. This could also help to port it to C++

Other than that you could also try to move the Arduino lib version of the Grove motor driver lib onto the teensy board and run the motor driver node from there. Or even add another Arduino mcu in addition to the pi and the teensy and run only the motor driver node on the Arduino.

However these are just some options that I can think of for now.

Russ76, 2022-11-20:
Were you able to use the DexterInd software without the Arduino IDE? I was trying to do that, but the Python code calls the Grove Driver node, so I was trying to compile that C++ module. Then I see this note in the DexterInd readme:
The GrovePi uses an Arduino to interface between the Raspberry Pi and the Grove Sensors, and comes programmed with a standard firmware. The firmware can be rewritten from the Raspberry Pi.
It's a bit confusing.

fjp, 2022-11-21:
I used the python motor driver lib from DexterInd without any mcu, only with the Raspberry Pi. It is also mentioned in the python module: https://github.com/DexterInd/GrovePi/blob/master/Software/Python/grove_i2c_motor_driver/grove_i2c_motor_driver.py#L5

You can see the schematic that I used for diffbot with the grove motor driver here: https://ros-mobile-robots.com/hardware_setup/electronics/ (make sure to select the diffbot tab)

[image]

As you can see the motor driver is directly connected to the Raspberry Pi. No Arduino used. Only the teensy for the encoders. Therefore the encoders.ino script.

> The GrovePi uses an Arduino to interface between the Raspberry Pi and the Grove Sensors, and comes programmed with a standard firmware. The firmware can be rewritten from the Raspberry Pi.

Sorry I couldn't find where this is mentioned. Can you please share a direct link?

Russ76, 2022-11-21:
Yes, I have mine wired up that way. I used the three Python files, and it is trying to work, but still throws an error. Like this:

pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- 0f 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ python3 motor_driver.py
Traceback (most recent call last):
  File "motor_driver.py", line 78, in <module>
    motor_driver = MotorDriver(address=0x0f)
  File "motor_driver.py", line 12, in __init__
    self.motor_driver = GroveMotorDriver(address=0x0f)
NameError: name 'GroveMotorDriver' is not defined
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ python3 motor_example.py
Traceback (most recent call last):
  File "motor_example.py", line 46, in <module>
    m = GroveMotorDriver(address=0x0f)
NameError: name 'GroveMotorDriver' is not defined
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ 

It's so close to working!

Russ76, 2022-11-21:
`pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- 0f 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ python3 motor_driver.py
Traceback (most recent call last):
  File "motor_driver.py", line 78, in <module>
    motor_driver = MotorDriver(address=0x0f)
  File "motor_driver.py", line 12, in __init__
    self.motor_driver = GroveMotorDriver(address=0x0f)
NameError: name 'GroveMotorDriver' is not defined
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ python3 motor_example.py
Traceback (most recent call last):
  File "motor_example.py", line 46, in <module>
    m = GroveMotorDriver(address=0x0f)
NameError: name 'GroveMotorDriver' is not defined
pi@raspberrypi:~/catkin_ws/src/grove_motor_driver/src$ 
`

Russ76, 2022-11-21:
I see the line is different: (line 24 and 26)
Class: motor_driver:
in one copy, and in the other, in your repo:
Class: GroveMotorDriver:
This is in "grove_i2c_motor_driver.py"

Russ76, 2022-11-21:
Ah, we got motors to work! Thanks

fjp, 2022-11-21:
You’re welcome, glad you got the motors working!

I'll close this issue in this case but feel free to re-open it in case there is another related issue with the grove motor driver. Thanks!

closed 2022-11-21 by fjp

Russ76, 2022-11-21:
Well, they run, but only up to about half power. That won't be enough for a tracked robot on carpet. I found something that may help the issue:
https://pypi.org/project/raspberry-i2c-tb6612fng/
Haven't tried it yet, but plan to. 
Thanks, Franz