ros-mobile-robots/diffbot #68: Motors don't turn

Russ76, 2022-11-29:
Cannot get the motors to run from the joystick. I checked, and it is publishing data to /diffbot/mobile_base_controller/cmd_vel. Also the /diffbot/wheel_cmd_velocities shows data from the joystick. But /motor_left and /motor_right do not show any data. If I send a command from CLI on that topic then the motors will run. I am running joy and teleop_twist_joy on the laptop. And I ran the launch file on robot called roslaunch diffbot_bringup bringup.launch

I am using the Python Grove motor controller routines. I don't have the Lidar tested yet; was saving that for after the driving was proven. The encoders seem to work correctly. The odom topic stays at zero, even when tracks are spun by hand. I tried using the rqt_robot_steering and it published but no response from program or motors. 

I'm using Raspi 4.0 with 4 Gig, Ubuntu 20.04 Mate and Noetic.

What should I try next to fix this robot?

pi@raspberrypi:~$ roswtf
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /diffbot/mobile_base_controller/joy_node:
   * /diffbot/mobile_base_controller/joy/set_feedback
 * /motor_driver:
   * /motor_left
   * /motor_right
 * /diffbot/diffbot_base:
   * /diffbot/measured_joint_states

Found 1 error(s).

ERROR The following nodes should be connected but aren't:
 * /diffbot/rosserial_base_controller->/diffbot/diffbot_base (/diffbot/encoder_ticks)

pi@raspberrypi:~$ roslaunch diffbot_bringup bringup.launch
... logging to /home/pi/.ros/log/ac5e6b76-6f8c-11ed-bbf4-d981b5fb025a/roslaunch-raspberrypi-2589.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:37209/

SUMMARY
========

PARAMETERS
 * /diffbot/debug/base_controller: True
 * /diffbot/debug/hardware_interface: False
 * /diffbot/encoder_resolution: 542
 * /diffbot/gain: 1.0
 * /diffbot/hardware_interface/joints: ['front_left_whee...
 * /diffbot/joint_state_controller/extra_joints: [{'name': 'rear_c...
 * /diffbot/joint_state_controller/publish_rate: 50
 * /diffbot/joint_state_controller/type: joint_state_contr...
 * /diffbot/mobile_base_controller/angular/z/has_acceleration_limits: True
 * /diffbot/mobile_base_controller/angular/z/has_velocity_limits: True
 * /diffbot/mobile_base_controller/angular/z/max_acceleration: 6.0
 * /diffbot/mobile_base_controller/angular/z/max_velocity: 8.0
 * /diffbot/mobile_base_controller/base_frame_id: base_footprint
 * /diffbot/mobile_base_controller/left_wheel: front_left_wheel_...
 * /diffbot/mobile_base_controller/linear/x/has_acceleration_limits: True
[... 112 lines cut ...]
 * /diffbot/motor_right [std_msgs/Int32] 1 publisher
 * /diffbot/wheel_cmd_velocities [diffbot_msgs/WheelsCmdStamped] 1 publisher
 * /diffbot/reset [std_msgs/Empty] 1 publisher
 * /diffbot/pid/left_motor/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /diffbot/pid/left_motor/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /diffbot/pid/right_motor/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /diffbot/pid/right_motor/parameter_updates [dynamic_reconfigure/Config] 1 publisher
 * /diffbot/encoder_ticks [diffbot_msgs/Encoders] 1 publisher
 * /diffbot/imu/data_raw [sensor_msgs/Imu] 1 publisher
 * /diffbot/joint_states [sensor_msgs/JointState] 1 publisher
 * /diffbot/mobile_base_controller/odom [nav_msgs/Odometry] 1 publisher
 * /diffbot/mobile_base_controller/parameter_descriptions [dynamic_reconfigure/ConfigDescription] 1 publisher
 * /diffbot/mobile_base_controller/parameter_updates [dynamic_reconfigure/Config] 1 publisher

Subscribed topics:
 * /rosout [rosgraph_msgs/Log] 1 subscriber
 * /diffbot/mobile_base_controller/joy/set_feedback [sensor_msgs/JoyFeedbackArray] 1 subscriber
 * /diffbot/mobile_base_controller/joy [sensor_msgs/Joy] 1 subscriber
 * /motor_left [std_msgs/Int32] 1 subscriber
 * /motor_right [std_msgs/Int32] 1 subscriber
 * /diffbot/joint_states [sensor_msgs/JointState] 1 subscriber
 * /diffbot/encoder_ticks [diffbot_msgs/Encoders] 1 subscriber
 * /diffbot/measured_joint_states [sensor_msgs/JointState] 1 subscriber
 * /diffbot/reset [std_msgs/Empty] 1 subscriber
 * /diffbot/mobile_base_controller/cmd_vel [geometry_msgs/Twist] 1 subscriber

fjp, 2022-11-29:
The issue seems to be that the grove_motor_driver node is [subscribed to `motor_left/right` without the `/diffbot/` prefix](https://github.com/ros-mobile-robots/grove_motor_driver/blob/de51e11d6989d787626631ff58086186227a5501/src/motor_driver.py#L80-L82).

Just to clarify, the `diffbot_base` is actually providing two "output" interfaces:

- motor commands and (high-level PID)
- rotational joint velocities (low-level PID)

The interface you are currently using with the encoder(-bno).ino script is the high-level PID approach and that one works with the motor commands, publishing setup is here:

https://github.com/ros-mobile-robots/diffbot/blob/e1a4a13089c1fa5341c1814479cac6e99bf677f7/diffbot_base/src/diffbot_hw_interface.cpp#L58-L60

The `diffbot_base` is launched with the `/diffbot` namespace

https://github.com/ros-mobile-robots/diffbot/blob/e1a4a13089c1fa5341c1814479cac6e99bf677f7/diffbot_base/launch/diffbot.launch#L13

So the options that should fix the issue you have is to 

- change the subscribed topic in the `grove_motor_driver` or 
- also launch the `grove_motor_driver` in the `/diffbot` namespace

> **Note**
> In case you use the encoder-bno.ino you have to tune the [high-level PID](https://ros-mobile-robots.com/packages/diffbot_base/high-level/)

Russ76, 2022-11-30:
Franz, you are right, now they run! Forward and backward looks good, however, the turning power is much too low. How do I boost the turning power? The angular boost?

fjp, 2022-11-30:
Great, glad that helped! You should be able to tune the [high-level PID controller](https://ros-mobile-robots.com/packages/diffbot_base/high-level/#pid-controller) using dynamic reconfigure through rqt. I am guessing the P gain can be turned up a bit.

Russ76, 2022-12-01:
I will work on that, thanks! My Python on laptop is messed up and won't run Rqt tools, but I can do it some other way. I tried boosting the angular from teleop_twist_joy but that didn't work.

Russ76, 2022-12-02:
I was not able to get the angular command to increase without also increasing the linear, and it is enough already. I tried adjusting all the variables. However, even with weak turning, two of us were able to drive the robot around successfully even on carpet. Maybe this will work with move_base and other software. 

The encoder topic is a problem. They aren't connected, as can be seen above. The Teensy code has topic "Encoder" and the Raspi code looks for topic "EncoderStamped" These are not the same! Did you get the package to work like this? How?

fjp, 2022-12-03:
> I was not able to get the angular command to increase without also increasing the linear, and it is enough already. I tried adjusting all the variables. However, even with weak turning, two of us [...]

Have you tried adapting the limits in the `diff_drive_controller` config? Maybe that could help (see also a [full configuration example of all values here](http://wiki.ros.org/diff_drive_controller#Complete_description)).

https://github.com/ros-mobile-robots/diffbot/blob/e1a4a13089c1fa5341c1814479cac6e99bf677f7/diffbot_control/config/diffbot_control.yaml#L45-L57

After that you most likely have to tune the PID gains again.

> The encoder topic is a problem. They aren't connected, as can be seen above. The Teensy code has topic "Encoder" and the Raspi code looks for topic "EncoderStamped" These are not the same! Did you get the package to work like this? How?

Right, that is an issue I introduced while switching from the high-level PID to the low-level PID approach. Sorry for that :-/ 

I'd recommend to update the `encoder.ino` / `encoder-bno.ino` to also use the `diffbot_msgs::EncodersStamped` message as it includes a time stamp, which might be useful.

Russ76, 2022-12-03:
Well, I tried to get that to work, and it compiles, and the topic connects, but there is no ticks data showing in the message!

fjp, 2022-12-04:
That is strange. Is the published topic from the mcu containing something when you inspect it, e.g. with something similar to `rostopic echo /diffbot/encoder_ticks`? If not then there seems to be an issue while creating the message. I am assuming that `diffbot_msgs::Encoders` is present as an (arudino) library. For PlatformIO I placed the header in [`diffbot_base/scripts/base_controller/lib/ros_lib/diffbot_msgs`](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_base/scripts/base_controller/lib/ros_lib/diffbot_msgs). The way that `diffbot_msgs::Encoders` is used for the new `base_controller` firmware is shown below:

https://github.com/ros-mobile-robots/diffbot/blob/e1a4a13089c1fa5341c1814479cac6e99bf677f7/diffbot_base/scripts/base_controller/lib/base_controller/base_controller.h#L523-L532

This is slightly different to how it is currently done in the `encoders-bno.ino`:

https://github.com/ros-mobile-robots/diffbot/blob/e1a4a13089c1fa5341c1814479cac6e99bf677f7/diffbot_base/scripts/encoders/encoders-bno/encoders-bno.ino#L95-L96

However, because it compiles I am guessing there is no issue here. 

What I would also try to verify is that `rosmsg show diffbot_msgs/Encoders` will correctly show the message. [^msg]

[^msg]: https://ros-mobile-robots.com/diffbot_msgs/#using-rosmsg

If all this doesn't help, I will try to find time near the end of the month to test this setup myself.

Russ76, 2022-12-05:
Yes, that block of code what what I needed. Now we got it working: encoders stamped has time, sequence, and ticks data! I had to add "diffbot" to the "encoder_diffbot.h" reference in the #include list, for the Arduino IDE to find the header file.  Now the only error in starting the main nodes on Raspi is complaining about no "joint states publishers", which I suppose is not important. Thanks, Franz!

Russ76, 2022-12-16:
Let the robots roam!

closed 2022-12-16 by Russ76