ros-mobile-robots/diffbot #74: No output for wheel_cmd_velocities? [Question]

Doctor-N0, 2023-02-13:
First of all: amazing repo! Absolutely great work!

I was launching the navigation and base via
```
roslaunch diffbot_base  diffbot.launch
roslaunch diffbot_navigation diffbot.launch
```
to get a more in-depth understanding of ros_control, specifically the hardware_interface's [write](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L163) function. Luckily, the functions published the diff_drive_controller's output (which it received via the [joint_velocity_commands_](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L183))

When I assign a drive command (e.g. angular velocity), the Gazebo Simulation works perfectly fine and the robot is spinning. However, the [/diffbot/wheel_cmd_velocities](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L186) yield only zero. 

Therefore, my question: Is this supposed to yield only zero values (see screenshot) or am I missing something?

Thanks so much for your help!

[image]

fjp, 2023-02-13:
Hi @Doctor-N0, the `diffbot.launch` from `diffbot_base` package is intended to be used by `minimal.launch`:

https://github.com/ros-mobile-robots/diffbot/blob/75a24b8d1da33ab600ccac45a713b667797a10ac/diffbot_bringup/launch/minimal.launch#L25-L31

This is used for running the real robot hardware. However usually a user shouldn't have to deal with those launch files. Instead you can find the recommended launch files for the real robot hardware in [this readme section](https://github.com/ros-mobile-robots/diffbot#navigating-and-mapping-on-the-real-robot).

Sorry, the [documentation here](https://ros-mobile-robots.com/packages/diffbot_base/high-level/#launch-file) is **not** up to date. Please refer to the [readme usage section](https://github.com/ros-mobile-robots/diffbot#navigation) in this case.

For simulation (navigation) you just can use this [lauch file command](https://github.com/ros-mobile-robots/diffbot#navigation):

``` 
roslaunch diffbot_navigation diffbot.launch
```

Or you can also do it with two separate launch commands like it is done for SLAM:

1. First launch Gazebo
    ```
    roslaunch diffbot_gazebo diffbot.launch 
    ```

2. Decide between SLAM or Navigation:
    ``` 
    roslaunch diffbot_slam diffbot_slam.launch slam_method:=gmapping
    ```
    In my most recent commit you can now also use navigation instead of slam with this command:
    ```
    roslaunch diffbot_navigation diffbot_navigation.launch
    ```

> Therefore, my question: Is it even POSSIBLE to run both, the navigation in gazebo AND the hardware_interface?

I think it should work and I tried and documented to some extent here: https://ros-mobile-robots.com/packages/diffbot_base/high-level/#simulation but you have to be careful to not launch two different controller managers and then loading the diff_drive_controller and others multiple times. So beware that this is not tested thoroughly and is for sure something that could be improved.

Doctor-N0, 2023-02-13:
Thanks for the detailed answer, that explains a lot!

Regarding Gazebo AND hardware_interface: My aim would be to run both in parallel, whereby in the write() function an OPC-UA connection is called. This way I can control a PLC, which in the first phase will only run virtually on another computer until it is finally transferred to the actual hardware.

If I get additional information concerning this topic, I will post it here.

fjp, 2023-02-13:
I would suggest to use RViz instead of "Gazebo AND the real robot hardware". RVIz is to visualize what the robot sees. Gazebo is to simulate your robot if you don't have the real robot hardware or want to do some testing without real hardware. To see the issue in more detail keep this image in mind:

[image]

For simulation with gazebo, diffbot uses the `gazebo_ros_control` package that works exclusively with ros control.

In case you try to run both, the real hardware and simulation (gazebo_ros_control package) will interfere with each other, because usually both receive the commands and then try to update the `JointStateInterface`. Then you would basically be running two diffbot's at the same time (one real, one simulated). So I guess you would have to tell gazebo not to listen to the joint commands and only update the current robot pose using for example the pose that amcl from the navigation package publishes. Or it might be possible to create your own plugin that listens to the joint states and implements the differential drive equations to update the robot pose accordingly. However, it seems not so straight forward to achieve. RViz is probably the right choice to see what the robot is actually doing.

To simulate your PLC, a simple node could do it if you don't need a visualization.

Depending on what you want to achive / which state variables you need, you could also think of adding a new function/class and use/instantiate it in the [main of the diffbot base](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_base.cpp#L5) and then call whatever you want periodically in the [control loop code](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_base.cpp#L31-L52) (where `diffbot.write()` is called periodically). If you need some state of the robot joints (e.g joint angles, velocity) you could also think of adding a new node that subscribes to those topics and then sends something to the PLC.

Of course it is also possible to put some logic directly in the write() method of the hardware interface but this method is anyway already doing two things (publishing a [wheel_cmd_msg](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L183) and a [motor commands](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L226-L227)). This is something I plan to refactor as it shouldn't do more than one thing.

Doctor-N0, 2023-02-14:
Thank you very much for this helpful reply. The idea to use rviz instead of gazebo is great. The other points are also very helpful, I will implement it that way. Thanks again!

closed 2023-02-14 by Doctor-N0