ros-mobile-robots/diffbot #79: wheel_radius problem

hsynklceng, 2023-02-23:
Hi,
I use diffbot package on my robot. I revised the parameters according to my own robot. But i have a problem. If I set the wheel_radius parameter to 1, all the values are correct, but if I set it to its real value for wheel_radius, the odom and desired_position datas are corrupted. 
How can i fix this problem?
Thank you

fjp, 2023-02-24:
Hi @hsynklceng thanks for using diffbot. Let me try to get more insight in your problem as I didn't encounter it so far with Diffbot or REMO. 

First of all, where are you setting the wheel_radius? And are you using the high-level or low-level PID approach? I am assuming the low-level PID approach which is based on the mcu firmware scripts in [diffbot_base/scripts/base_controller](https://github.com/ros-mobile-robots/diffbot/tree/noetic-devel/diffbot_base/scripts/base_controller).

> If I set the wheel_radius parameter to 1, all the values are correct, but if I set it to its real value for wheel_radius, the odom and desired_position datas are corrupted.

Regarding the wheel radius definition, Diffbot and REMO use also use 0.0325 m (3.25 cm), defined here:

https://github.com/ros-mobile-robots/diffbot/blob/8d1f5b70d946ac0f3c50c4158062abcac6b513ad/diffbot_control/config/diffbot_control.yaml#L30-L34

And I didn't encounter any odom issues. A value of 1 would mean a wheel with a diameter of 2 meter. Which value did you set and which desired_position calculation are you referring to? A link to the code location would help me to give better support.

## Other adaptations

It is also a good idea to update the wheel radius in your URDF. For example [here (diffbot)](https://github.com/ros-mobile-robots/diffbot/blob/a8057b4944792d7ba0dcd8e03f2156aaf41ffe2f/diffbot_description/config/diffbot/front_wheel.yaml#L2) and [here (REMO)](https://github.com/ros-mobile-robots/remo_description/blob/main/config/remo/front_wheel.yaml#L2), but this should only be relevant for simulation. The real robot reads the [`wheel_radius` from the parameter server](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L33).

One thing that you also have to adapt - in case you have different motors/encoders - is to set the number of encoder ticks for one full revolution. The important file where to set it is here:

https://github.com/ros-mobile-robots/diffbot/blob/ea002d04e8e7c4f7261e5f27fb03c7c5962a30dc/diffbot_base/config/base.yaml#L5

The one defined in the diffbot_base_config should be only relevant [for testing](https://github.com/ros-mobile-robots/diffbot/blob/61f25c806501b3a269914b09f606d121ac80b7b8/diffbot_base/scripts/base_controller/test/encoders/test_encoders.cpp), but it is a good idea to also set it correctly:

https://github.com/ros-mobile-robots/diffbot/blob/8d1f5b70d946ac0f3c50c4158062abcac6b513ad/diffbot_base/scripts/base_controller/lib/config/diffbot_base_config.h#L8

Another important thing is the PID tuning:

https://github.com/ros-mobile-robots/diffbot/blob/8d1f5b70d946ac0f3c50c4158062abcac6b513ad/diffbot_base/scripts/base_controller/lib/config/diffbot_base_config.h#L18-L20

Let me know if you are not using the low-level PID approach. Then the PID tuning looks a bit different.

hsynklceng, 2023-02-24:
firstly, thanks for answer

> First of all, where are you setting the wheel_radius? And are you using the high-level or low-level PID approach? I am assuming the low-level PID approach which is based on the mcu firmware scripts in

I set on diffbot_control.yaml  and yes i am using low-level PID on mcu. By the way, my wheel radius is 7.5 cm(0.075).

> And I didn't encounter any odom issues. A value of 1 would mean a wheel with a diameter of 2 meter. Which value did you set and which desired_position calculation are you referring to? A link to the code location would help me to give better support.

I solved the odometry problem, it was caused by a code error in my mcu. And If I add the parameter `publish_wheel_joint_controller_state: True` to diffbot_control.yaml, the topic containing the desired_position will be active.

fjp, 2026-10-08:
Closing because there has been no activity since 2023. If you still see this with the current code, please reopen it with your wheel_radius value and whether you use the low-level PID on the Teensy.

closed 2026-10-08 by fjp