ros-mobile-robots/diffbot #80: Question about Motors, minimum accomplishable RPM

altineller, 2023-03-21:
Hello, 

I am using the same type of motors, and a custom made driver board for my robot. Everything is working well, but there is a minor problem: These motors have a high deadzone, and the PID algorithm is having a hard time - controlling them at low speeds, such as < 5RPM. The problem gets worse, as the robot weight increases. :(

I have tried many things, such as lowering the PWM frequency, (the lower the pwm frequency, the less deadzone we have, but at frequencies like 100hz, the system generates electrical noise. Another trick that I tried is to read encoders, and current feedback and change pid constants when the motor is not moving, which does work - but then leads to stability problems in the PID algorithm.

What was your experience on this problem? What was your lowest observed wheel RPM? I understand you are doing the PID in the teensy, and not doing something non-standard?

Best Regards,
Can Altineller

fjp, 2023-03-23:
Hi Can,

I've experienced the same issues with these motors. Here are some videos showing the tests I did when I started. The first one shows the deadzone issue you mention:

- [DiffBot #12 Trouble driving straight with simple ROS Control Hardware Interface](https://youtu.be/chUPeWXtim4)
- [DiffBot ROS Hardware Interface with high level PID Controller #16](https://youtu.be/fdn5Mu0Qhl8). I ended up using F=0.8, P=0.35, I=0.5, D=0.01, which you can also see at the top of this [terminal output](https://youtu.be/IcYkQyzUqik?t=21)

After switching from the high level PID (running on the SBC in the ROS hardware interface) to the low level PID (running on the Teensy mcu) and using the sturdy REMO 3D printed parts, I think the behavior got slightly better but I am still using only two PID controllers (both with the same gains) and no further tricks. What might also help is to use the [gain trim model](https://github.com/ros-mobile-robots/diffbot/blob/noetic-devel/diffbot_base/src/diffbot_hw_interface.cpp#L169-L175) that's also used on the [duckiebots](https://ros-mobile-robots.com/packages/diffbot_base/pid/#gain-trim-model). However, this all doesn't really improve the issue with low speeds and I didn't investigated the issue further yet.

Where did you lower the PWM frequency and which value do you use? As far as I know I left the [default](http://adafruit.github.io/Adafruit_Motor_Shield_V2_Library/html/class_adafruit___motor_shield.html#ac2bd0e50c89bdd41996939f1c4a43c3f) of [1600 Hz](https://github.com/ros-mobile-robots/diffbot/blob/8d1f5b70d946ac0f3c50c4158062abcac6b513ad/diffbot_base/scripts/base_controller/lib/motor_controllers/adafruit_feather_wing/adafruit_feather_wing.h#L49) (when using the low level PID approach). I am not sure what value is set when I used the high level PID and the [grove motor driver](https://github.com/ros-mobile-robots/grove_motor_driver/blob/main/src/grove_i2c_motor_driver.py#L29).

Best Regards
Franz

altineller, 2023-03-27:
Hi Franz,

I have read the relevant parts of the code. I understand you are extending the control_toolbox for PID. And I also understand you are using 1.0 for gain and motor constant as the default, and F=0.8 is an pid output multiplier, i.e. output transfer function.

I will give a try to the duckietown trim model, which seems practical.

These motors work best from 50-100hz, with fast decay model. Fast decay model means when the pwm is at 0, the driver allows the two terminals of motor be shorted, i.e. brake mode, giving it better precision, lower deadzone, but higher power consumption.

I am working on a custom made controller board that I made myself, that features an arm cortex mcu, which has very advanced pwm generation features. I believe your PWM frequency of 1600hz, is inefficient for these motors.

I have tried to make a digital 'vitesse' that changes the PWM frequency, when the motor is not moving it starts from 10hz, and as the duty cycle increases, it will increase the pwm frq to 25hz, 50hz and 100hz. However this idea failed, as the efficiency of motors are different for different pwm frequencies, it introduces an instability to PID algorithm. With my current setup, I can not use different frequencies for left and right motor, (i have to wire up the second motor, to a different pwm generator on my arm mcu, which i will try with another iteration) - so I will attempt again at another time.

Best Regards,
C.