ros-mobile-robots/diffbot #91: Only one wheel is controlled while keyboard_teleop

DidarSeyidov, 2024-05-30:
When I control motors one motor is controlled, other one is always running even to command  a stopping

fjp, 2024-09-09:
Hi @DidarSeyidov are you still facing this issue and can you share your setup in a bit more detail and maybe also some log files?

DidarSeyidov, 2024-11-05:
Hi. Actually I solved that problem, but I cannot solve this issue.

[ERROR] [1730793200.753187948]: Global Frame: odom Plan Frame size 353: map
[ WARN] [1730793200.753271778]: Could not transform the global plan to the frame of the controller
[ERROR] [1730793200.753361173]: Could not get local plan
^C[ INFO] [1730793200.852993851]: Got new plan
[ERROR] [1730793200.853108113]: Extrapolation Error: Lookup would require extrapolation 0.020891836s into the future.  Requested time 1730793200.754234076 but the latest data is at time 1730793200.733342171, when looking up transform from frame [odom] to frame [map].

 Could you help me with this stuff

fjp, 2024-11-05:
Hi @DidarSeyidov this was reported once in [this discussion](https://github.com/orgs/ros-mobile-robots/discussions/56#discussion-3824872). It seems that the `transform_tolerance` did help in that case (mention in this [comment](https://github.com/orgs/ros-mobile-robots/discussions/56#discussioncomment-2036542)).

> My guess is that the parameter settings were tuned for a quite powerful hardware. I am pretty sure, commit https://github.com/ros-mobile-robots/diffbot/commit/9e155597e7ce1d0db729822e3e48c04c2128c13c introduced the issue you are experiencing. In this commit I tuned amcl for simulation but didn't test thoroughly on the real hardware afterwards. I would suggest adapting these values (especially [transform_tolerance](https://github.com/ros-mobile-robots/diffbot/blob/a018318c368fc3d637aec62c9e5d1ad1efde7ccd/diffbot_navigation/launch/amcl.launch#L25) from 0.1 back to 0.5) until you get rid of the error. If I find time I will try to change these too and find a good mix between simulation and hardware. Would be interesting if changing these values can solve your issues. And thanks for reporting your results so far!
> 
> Here is the [link to amcl parameters](http://wiki.ros.org/amcl#Parameters) and what transform_tolerance is doing:
> 
> ~transform_tolerance (double, default: 0.1 seconds)
> Time with which to post-date the transform that is published, to indicate that this transform is valid into the future.

DidarSeyidov, 2024-11-06:
Thank you @fjp. That solved my issue. One more thing I cannot access to mesh files, couldn't you upload them to google drive or something like that. remo_description repository is not found too. Please help me this stuff.

fjp, 2024-11-07:
@DidarSeyidov happy to hear it did solve your issue as well and thanks for letting me know!

Thanks for your interest in the project. I understand you're looking for the mesh files from the `remo_description` repository. Unfortunately, I had to make this repository private due to bandwidth limitations with Git LFS - the frequent downloads were consuming the entire organization's bandwidth quota and Github doesn't automatically reset the quota after each month, making it impossible for anyone (including myself) to clone the repositories.

Instead of hosting these files on Google Drive, I've made them available for a small one-time fee of $10 through [Gumroad](https://fjp.gumroad.com/l/GnMpU). This helps cover the costs of the professional design work that went into creating these files. 

You're welcome to continue using all the software components of diffbot for free! And if you prefer, you can also design your own custom robot chassis and model it in URDF with the https://github.com/ros-mobile-robots/mobile_robot_description while still using our open-source software stack.

I hope you understand that taking the repo private was necessary. Whether you choose to use the REMO design or create your own, I'm here to help if you have any questions!

fjp, 2026-10-08:
Closing, since the fix discussed above solved it. Thanks for reporting back, @DidarSeyidov!

closed 2026-10-08 by fjp