ros-mobile-robots/diffbot #92: Project directory structure after catkin_ws creation

ashwanthrkumar, 2024-07-08:
Can anyone share the project directory structure after catkin_ws creation

fjp, 2024-09-09:
Hi @ashwanthrkumar sorry for the late answer. My structure of the catkin workspace (called `ros_ws`) looks like the following:

```
~/ros_ws$ tree -d
.
├── build
│   ├── CMakeFiles
│   │   ├── 3.16.3
│   │   │   ├── CompilerIdC
│   │   │   │   └── tmp
│   │   │   └── CompilerIdCXX
│   │   │       └── tmp
│   │   ├── CMakeTmp
│   │   ├── clean_test_results.dir
│   │   ├── download_extra_data.dir
│   │   ├── doxygen.dir
│   │   ├── run_tests.dir
│   │   └── tests.dir
│   ├── atomic_configure
│   ├── bin
│   ├── catkin
│   │   └── catkin_generated
│   │       └── version
│   ├── catkin_generated
│   │   ├── installspace
│   │   ├── metapackages
│   │   │   └── diffbot_robot
│   │   └── stamps
│   │       └── Project
│   ├── diffbot
│   │   ├── diffbot_base
│   │   │   ├── CMakeFiles
│   │   │   │   ├── control_msgs_generate_messages_cpp.dir
│   │   │   │   ├── control_msgs_generate_messages_eus.dir
│   │   │   │   ├── control_msgs_generate_messages_lisp.dir
│   │   │   │   ├── control_msgs_generate_messages_nodejs.dir
│   │   │   │   ├── control_msgs_generate_messages_py.dir
│   │   │   │   ├── control_toolbox_gencfg.dir
│   │   │   │   ├── control_toolbox_generate_messages_cpp.dir
│   │   │   │   ├── control_toolbox_generate_messages_eus.dir
│   │   │   │   ├── control_toolbox_generate_messages_lisp.dir
│   │   │   │   ├── control_toolbox_generate_messages_nodejs.dir
│   │   │   │   ├── control_toolbox_generate_messages_py.dir
│   │   │   │   ├── controller_manager_msgs_generate_messages_cpp.dir
│   │   │   │   ├── controller_manager_msgs_generate_messages_eus.dir
│   │   │   │   ├── controller_manager_msgs_generate_messages_lisp.dir
│   │   │   │   ├── controller_manager_msgs_generate_messages_nodejs.dir
│   │   │   │   ├── controller_manager_msgs_generate_messages_py.dir
│   │   │   │   ├── diff_drive_controller_gencfg.dir
│   │   │   │   ├── diffbot_base.dir
│   │   │   │   │   └── src
│   │   │   │   ├── diffbot_base_gencfg.dir
│   │   │   │   ├── diffbot_hw_interface.dir
│   │   │   │   │   └── src
│   │   │   │   ├── trajectory_msgs_generate_messages_cpp.dir
│   │   │   │   ├── trajectory_msgs_generate_messages_eus.dir
│   │   │   │   ├── trajectory_msgs_generate_messages_lisp.dir
│   │   │   │   ├── trajectory_msgs_generate_messages_nodejs.dir
│   │   │   │   └── trajectory_msgs_generate_messages_py.dir
│   │   │   ├── catkin_generated
│   │   │   │   ├── installspace
│   │   │   │   └── stamps
│   │   │   │       └── diffbot_base
│   │   │   └── docs
│   │   ├── diffbot_bringup
│   │   │   ├── CMakeFiles
│   │   │   │   └── _catkin_empty_exported_target.dir
│   │   │   └── catkin_generated
│   │   │       ├── installspace
│   │   │       └── stamps
[... 281 lines cut ...]
    │   │   ├── remo
    │   │   │   ├── camera_mount
    │   │   │   ├── caster
    │   │   │   ├── deck
    │   │   │   └── lidar_platform
    │   │   └── sensors
    │   ├── rviz
    │   └── urdf
    │       └── include
    └── rplidar_ros
        ├── launch
        ├── rviz
        ├── scripts
        ├── sdk
        │   ├── include
        │   └── src
        │       ├── arch
        │       │   ├── linux
        │       │   ├── macOS
        │       │   └── win32
        │       └── hal
        └── src

368 directories
```

fjp, 2024-09-09:
If there are any issues or additional questions, feel free to re-open this issue or another one.

closed 2024-09-09 by fjp