ros-mobile-robots/diffbot #96: Username? Password ???

innomin0409, 2025-02-06:
I downloaded Git Source!
      git clone https://github.com/fjp/diffbot.git
and
I was typing "vcs ~"
      vcs import < src/diffbot/diffbot_dev.repos

but
The system requested "Username" and "Password"
   Username for 'https://github.com': AAA
   Password for 'https://AAA@github.com': 

so 
What are Username and Password?

fjp, 2025-02-06:
Hi @innomin0409 thanks for reporting the issue. I assume it is because I made the remo_description repo private. But can you please provide the whole error message that you see?

fjp, 2025-02-06:
I had to make the remo_description repo private because otherwise the huge amount of downloads each months would freeze the whole cloning of any repo in this organization, because of the limits GitHub has on it's free organization plans. So usually you can only get access to remo_description after [obtaining a license](https://fjp.gumroad.com/l/xqfpp). Since you are the first to report this issue, I've given you direct access to the private remo_description repo. You should have received an invitation, which you can accept. After that it should be possible to clone it.

fjp, 2026-10-08:
remo_description is public again, so `vcs import` no longer asks for a username and password. The STL meshes aren't in the public repo: the files there are empty placeholders, and the real ones come with the [Remo files on Gumroad](https://fjp.gumroad.com/l/GnMpU), downloaded with `scripts/download_assets.py` (see the [remo_description README](https://github.com/ros-mobile-robots/remo_description#stl-mesh-files)).

closed 2026-10-08 by fjp